ni/measurement-plugin-labview #165: Improve generator's handling of cases where conflicting measurement / resources exist

aepete, 2023-01-25:
# Bug Report

We have a bit of logic in `Valid Measurement Service Name.vi` that helps prevent problematic collisions, but it only checks against libraries present in the active project and against a (stale) handful of reserved names.  There are several reasonable workflows whereby the generator allows the user to proceed despite an eminent & potentially destructive collision:
- If the user has a colliding library on disk but not in the project, their disk content will be overwritten by the generation process, without ever having provided the user with any notice.  (Due to the destructive behavior, I'd think we should either fail entirely or provide a warning / opt-out.)
- If the user has a colliding build specification in their project (a build spec with the same name as their pending measurement plug-in), the generation process throws an error -4500 dialog.  The generator dialog continues running, too, which I didn't expect.
[image]
- We should probably update the stale list of reserved library names, in light of changes since `Valid Measurement Service Name.vi` was last updated.  This bit of code seems to be providing protection against the unlikely circumstance that the user tries to assign a library name that would collide with one of the libraries that MeasurementLink already packages and delivers within vi.lib, e.g. `MeasurementLink Discovery Client.lvlib`, `MeasurementLink Measurement Server.lvlib`, etc.

## Repro or Code Sample

Here is the current state of the name-checking VI, with problematic portions highlighted:
[image]
- Orange circled area only checks for collisions within the project, and it fails to check on disk.  There is also no logic to check against build spec names that might already exist in the project.
  - Note that we'd likely employ better code re-use here by calling `Get Measurement Library.vi`, instead of a series of VI Server nodes dealing with project resources.
- Red circled area contains a stale set of reserved names.

## Expected Behavior

## Current Behavior

## Possible Solution

## Context

## Your Environment

* OS & Device: [e.g. macOS, Windows, Linux] on [Mac, PC]
* `ni-measurement-service` version [e.g. 0.12.3]
* MeasurementLink version [e.g. 23.0.0d123]
* Python version [e.g. 3.9]

aepete, 2023-02-07:
Incidentally, this bug involves the same area of code as #168 ...

dixonjoel, 2025-01-09:
We haven't received any feedback on this and aren't like to get to it anytime soon.

closed 2025-01-09 by dixonjoel