ni/measurement-plugin-labview #168: Poor handling when user puts backslash(es) into pending measurement name in G generator

aepete, 2023-02-06:
# Bug Report

In most cases, if the user enters an unsupported character (e.g. dollar sign, hashtag, at-symbol, etc.), they are stopped cold and given a helpful error.  But if a backslash is the only bad character in a name, the user is given a different error, and they are prompted to save the measurement library with a different name.  After that, they get another error.  And in the end, their disk gets dirtied, but they don't get anything in their project.

## Repro or Code Sample

[image]
[image]
[image]
- Notice that the generator has actually put the default template content onto disk, in a subhierarchy that reflects the slash-delimited measurement name
- Even if the user provides a valid library name when offered in this dialog, the measurement generator code still errors, as follows, because it is built to expect to find the library with having the name provided in the dialog:
[image]

## Expected Behavior

The generator should reject names with backslashes before it gets so far as to create directories on disk, copy the template content, etc.

## Current Behavior

## Possible Solution

## Context

## Your Environment

* OS & Device: [e.g. macOS, Windows, Linux] on [Mac, PC]
* `ni-measurement-service` version [e.g. 0.12.3]
* MeasurementLink version [e.g. 23.0.0d123]
* Python version [e.g. 3.9]

DelpireNI, 2023-02-06:
> In most cases, if the user enters an unsupported character (e.g. dollar sign, hashtag, at-symbol, etc.), they are stopped cold and given a helpful error.

@aepete I have not seen this in my testing today. I have been able to create measurements with special characters:
[image]

aepete, 2023-02-06:
I also noticed that the dialog's handing of leading and trailing whitespace in the name is non-ideal:
- leading whitespace actually generates the measurement, with an awkward-looking name that actually contains those leading spaces, as shown:
[image]

- trailing whitespace fails to generate (good..), but the error is not very helpful.
[image]

The dialog should either alert the user that leading & trailing whitespace is unsupported immediately following text-entry -or- fail with a more helpful error when the user clicks the create button.

aepete, 2023-02-06:
@DelpireNI interesting.  I suspect those are being allowed windows considers them valid filenames.  But now that I check again, all the broken names that I had tested with also included back characters such as the at-sign or question mark, which windows rejects (.. thus the dialog rejects..).  Thanks for pointing this out though; it's evident that the dialog is currently allowing some characters that we may not want to support.  FYI @rfriedma .

aepete, 2023-02-06:
I've updated the bug description to be clearer about which particular characters cause issues.

aepete, 2023-02-06:
FTR, here are the characters that are rejected by Windows:
[image]

Seems that the current implementation of the dialog is consistent with this, except for the backslash, for which it goes through a very long-winded failure path.

aepete, 2023-03-06:
Closing as fixed.