ni/measurement-plugin-labview #174: Could not open sfp file in DCPower example

LawsonGu, 2023-02-13:
When I was trying to run the [DCPower example](https://github.com/ni/measurementlink-labview/tree/main/Source/Example%20Measurements/NI-DCPower%20Source%20DC%20Voltage), I open the NIDCPowerSourceDCVoltage.instudioproj and then double click run the sfp file. Something stop me. 
[image]

My InstrumentStudio version and setting of plugin
[image]
[image]

pbirkhol-ni, 2023-02-13:
@LawsonGu, did you install MeasurementLink 23.0? That is required to use this example.

LawsonGu, 2023-02-14:
> @LawsonGu, did you install MeasurementLink 23.0? That is required to use this example.

Yes, download from argo
[image]

closed 2023-02-14 by pbirkhol-ni