ni/measurement-plugin-labview #194: Measurement service starts successfully when discovery service can't be contacted

jasonmreding, 2023-04-06:
# Bug Report

When developing a measurement service and testing it, it's typical to use the development environment to start the service manually. However, for the service to be used by the rest of the system, the discovery service must already be started and the service must be able to register itself with it successfully. Currently, if the discovery service isn't started when the service starts, the service still appears to start and run successfully. However, later attempts to access the service from a client will always fail. When you manually stop the service, only then do you see an error reported about not being able to contact the discovery service.

## Repro or Code Sample

1. If the discovery service is already started, kill it from task manager.
2. Manually start a measurement service using the Run Service.vi in the LV IDE. Notice no error is generated, and the service continues to run.
3. Try to access the measurement from TestStand or InstrumentStudio. Notice the measurement isn't found unless you happen to have already registered a .serviceconfig file for the service.
4. Manually stop the measurement service in the LV IDE. You will now get a popup error about not being able to contact the discovery service.

## Expected Behavior

I would expect trying to run the service in this scenario would automatically fail and popup an error message without requiring manual intervention to stop it.

## Current Behavior

User must manually stop the measurement service before receiving any indication that the service isn't usable.

## Context

The current behavior wastes time during development if you forget to start the discovery service. It's not uncommon to need to kill the discovery service during development as you're working out the bugs or abort code execution so that cleanup functions don't run.

dixonjoel, 2023-04-06:
@jasonmreding which versions of the .vip packages are you using? We corrected this behavior with a recent PR.
#181

jasonmreding, 2023-04-06:
[image]

Dang, I browsed through recent commits but didn't spot anything that looked like it was this issue. I'll go ahead an close out the issue then since it looks like it will be addressed if I upgrade to the latest release.

jasonmreding, 2023-04-06:
Closing as duplicate.

closed 2023-04-06 by jasonmreding