ni/measurement-plugin-labview #195: How to transfer a manual test VI into MeasurementLink valid VI

LawsonGu, 2023-04-12:
Hi, team
Our customers requirement is: they have a lot of manual test Vis. They want to register them in MeasurementLink as a service so that run in TestStand or play in InsturumentStudio. Here is two questions:
1. Even they have read the manual from ni.com, they still don't know how to transfer their 64-bit VIs into MeasurementLinks' valid Vi. Is there any learning material like document or video I can share with customers. In this case I should say that the manual doen't cover this part.
2. Even I have told them MeasurementLink doen't support 32-bit LabVIEW. But customers insist that reusing their 32-bit manual test VIs. How to work around? One thing I think work is use Python to call up these 32-bit VIs. But seems a little slow. Any better method?

santhoshramaraj, 2024-06-05:
@LawsonGu , I contacted over Teams to discuss a tool that could help you with this requirement.

dixonjoel, 2024-10-18:
Other than our current documentation, we won't make any changes for this at this time.

closed 2024-10-18 by dixonjoel