ni/measurement-plugin-labview #485: MeasurementLink Client LabVIEW example frequently crashes.

ccaltagi, 2024-02-20:
# Bug Report

## I am trying to run the shipping MeasurementLink Client example by first running `Run Service.vi` in `Service.lvproj` then `Run Client.vi` in `Client.lvproj`, but this crashes very frequently.
I am attaching the crash reports.

Run the example from measurementlink-labview-examples-2.0.0.1\MeasurementLink Client

## Expected Behavior

The example should not crash.

## Current Behavior

The example crashes when running the client.
Attaching the stack trace.

## Context

Evaluation of adding MeasurementLink support to FlexLogger.

## Your Environment

* OS & Device: Windows 11
* `ni-measurement-service` version 2.0.0.1
* MeasurementLink version 2024 Q1
* LabVIEW gRPC Library 1.2.0.1

[8b8a5ea8-93d6-4c56-a520-276ba41bf2fc.zip](https://github.com/ni/measurementlink-labview/files/14350574/8b8a5ea8-93d6-4c56-a520-276ba41bf2fc.zip)
[3401af17-584c-45dd-9e00-8d4033496da2.zip](https://github.com/ni/measurementlink-labview/files/14351222/3401af17-584c-45dd-9e00-8d4033496da2.zip)

dixonjoel, 2024-02-20:
@ccaltagi I noticed you have the 1.2.0.1 version of LabVIEW gRPC Library. We depend on the 1.0.1.1 version (see the assets in our release [here](https://github.com/ni/measurementlink-labview/releases/tag/v2.0.0.1)). I'd suggest trying with that version to see if it still crashes. I ran the client several times on my machine here without crashing.

ccaltagi, 2024-02-20:
Thanks! Downgrading to gRPC 1.0.1.1 fixed the crash.

dixonjoel, 2024-02-20:
@CPattar-NI The 1.2.0.1 release seems to have broken this. We looked through the changes and suspect maybe https://github.com/ni/grpc-labview/pull/335 could be the culprit. Would you take a look at this?

nischalks, 2024-03-13:
This is now fixed and is available as part of this release https://github.com/ni/grpc-labview/releases/tag/v1.2.2.1

dixonjoel, 2024-06-04:
Closing as the fix has been verified.

closed 2024-06-04 by dixonjoel