ni/measurement-plugin-labview #559: Reference Leak

sabthagiri6, 2024-07-08:
# Reference of the libraries are not closed properly

The reference of all the library are opened, the reference is indexed through a for loop until a library of the same name is found. If library of same name is found, many reference left unclosed which ends up in reference leak.

## Repro or Code Sample

[image]

## Expected Behavior

The reference of rest of the libraries should be closed.

## Possible Solution

Get the index of the reference, and retain the reference which is required and close rest of the reference.

ni-vsts-admins, 2024-07-08:
✅ Successfully linked to Azure Boards work item(s):
- [User Story 2795144](https://dev.azure.com/ni/94b22d7b-ad7b-4f5e-88f0-867910f91c94/_workitems/edit/2795144)

varshini-senthil, 2025-08-29:
> I am going to close this issue. I did an audit of the code, and the VI shown above seems to close out all references it is supposed to. The VI itself should not close the refnum to the library it is looking for as that is the point of the VI. It is up to the callers of this VI to ensure they close the returned refnum. I also audited all callers and they all do close the refnum when they are done with it so I don't believe there are any leaks. If I have missed something, feel free to reopen the issue.

Hi @jasonmreding,

I wanted to kindly share a scenario that might help with understanding the behavior of this VI, which closes library references obtained from the `Get All Descendants` invoke node.

In a scenario where the invoke node returns references to multiple libraries - for example, [A, B, C, D] - and the VI is searching for a reference to library B, the `for` loop exits as soon as it finds a match (in this case, during the second iteration). As a result, references to libraries C and D are not processed or closed, which could potentially lead to a reference leak.

Could we please consider this case and possibly revisit the issue?

Thank you so much for your time and consideration!

jasonmreding, 2025-09-03:
I missed the stop terminal stopping the loop short. Yes, I'll reopen the issue.