ni/measurement-plugin-labview #594: LabVIEW crash with update of gRPC package version to 1.2.5.1

varshini-senthil, 2024-09-17:
# Bug Report

LabVIEW crashes when running `Enumerate Services.vi` under the Discovery client wrappers, with update of gRPC package version to 1.2.5.1 from 1.0.1.1.

## Repro or Code Sample

- Install the packages provided in the latest release in our repo: [[Measurement Plug-In SDK for LabVIEW 3.0.0.4](https://github.com/ni/measurement-plugin-labview/releases/tag/v3.0.0.4)](https://github.com/ni/measurement-plugin-labview/releases/tag/v3.0.0.4)
- Run the `Enumerate Services.vi` under the Discovery client wrapper library `NI Discovery V1 Client.lvlib`
- LabVIEW crashes with the following error code:

## Expected Behavior

## Current Behavior

## Possible Solution

## Context

## Your Environment

* OS & Device: [e.g. macOS, Windows, Linux] on [Mac, PC]
* Measurement Plug-In SDK Service version [e.g. 3.0.0.3]
* InstrumentStudio version [e.g. 24.3.0d123]
* LabVIEW version [e.g. 2021 SP1]

varshini-senthil, 2024-09-17:
This issue caused by the update of gRPC LabVIEW packages to version 1.2.5.1, affects the `Measurement Plug-In Client Generator` tool, as the Client generator uses the `Enumerate Services` API to get the list of available V2 measurement services.

cc: @dixonjoel @jasonmreding

jasonmreding, 2024-09-23:
See https://github.com/ni/grpc-labview/issues/384.

closed 2024-10-22 by Avinash2Suresh