ni/measurement-plugin-labview #660: Support to automate Run Service.vi

bharathirm11, 2025-07-25:
## Problem to Solve

We’ve developed 15+ measurement plugins for device validation, However, we have not yet implemented static registration or executable builds for these measurements since the measurement procedures are not finalized for our new product. To run or debug a plugin, users must manually navigate to the file directory and launch Run Service.VI for each one.

This manual process is inefficient, and there's no API support to start/stop services programmatically, either individually or in batch.

As a workaround, we modified the Measurement SDK to change the "Stop" control from Latch to Switch, enabling basic programmatic control.

[image]

## Proposed Solution

- Provide an API to start/stop measurement services without user popups programmatically.
- Add an API to list currently running services.

jasonmreding, 2025-07-25:
> Add an API to list currently running services.

This can already be done via the [discovery service API](https://github.com/ni/ni-apis/blob/main/ni/measurementlink/discovery/v1/discovery_service.proto#L39). There is a wrapper VI around a call to that API installed to `vilib` called `Enumerate Services.vi` which you can use. However, it is not currently on the palette as you generally don't need that to implement a measurement service. Does this meet your needs?

jasonmreding, 2025-07-25:
BTW, we have used the above to successfully start, test, stop services in automated test suites.

jasonmreding, 2025-07-28:
> I have another request is to expose the "hide window" flag in the measurement user settings for the developer environment. 

That is a fair criticism. You are welcome to file a separate issue/bug for that and/or submit a PR for that change if it is something that is hindering you.

> Is there an API available Discovery Service or a notifier(named as a service class) to stop the service?

There is currently no programmatic API for stopping the service outside of using LV programming constructs to signal the service to stop (like the user event described above). With that said, there is nothing stopping you from defining your own grpc service API for stopping the server using grpc-labview and adding that to your server implementation for each measurement. If you build your services into executables, you can also use console commands like `taskkill` to stop them in place of a formal programmatic API.

jasonmreding, 2025-08-13:
The following is the AI generated overview for "github submit PRs via fork rather than branch".

------------------------------------------------------------------------------------------------------------

Submitting Pull Requests (PRs) via a fork on GitHub is a common workflow, especially in open-source projects or when contributing to repositories where direct write access is not granted. This method contrasts with creating a branch directly within the repository if one has write access.

The process of submitting PRs via a fork typically involves the following steps: 

**Fork the Repository:**
Create a personal copy (fork) of the target repository under your own GitHub account. This creates a separate repository that you control.

**Clone Your Fork:**
Clone your newly created fork to your local machine. This allows you to work on the code locally.

**Create a Feature Branch:**
Within your local clone of the fork, create a new branch for the specific changes or feature you are implementing. This keeps your changes isolated.

**Make and Commit Changes:**
Implement your desired changes within this feature branch and commit them.

**Push to Your Fork:**
Push the feature branch containing your changes from your local machine to your fork on GitHub.

**Create a Pull Request:**
Navigate to the original (upstream) repository on GitHub. GitHub typically detects the new branch in your fork and provides a prompt to create a pull request. Alternatively, you can manually create a pull request, selecting your fork and the relevant branch as the source, and the upstream repository's target branch as the destination.

**Review and Merge:**
The maintainers of the original repository will review your pull request, potentially request changes, and ultimately merge it into the main project if it meets their criteria. 

This "fork and pull request" workflow is particularly useful for external contributors as it allows them to propose changes without needing direct write access to the main repository, promoting a more controlled and collaborative contribution environment.