ni/measurement-plugin-labview #664: Measurement plug-in SDK Examples incorrect folder structure

bharathirm11, 2025-08-14:
# Bug Report

Measurement plug-in SDK Examples folder is incorrect in 3.4.0.1 & 3.5.0.1. 

[image]

## Repro or Code Sample

Install latest vipm pkg of Measurement plug-in SDK Examples

## Expected Behavior

In ver 3.3.1.1 and older,

[image]

bkeryan, 2025-08-15:
I confirmed with 7-zip that the package layout is different.

3.3.0.1

[image]

3.3.1.1

[image]

3.4.0.1

[image]

3.5.0.1

[image]

Blahpapto1, 2025-08-19:
I think its because of this. I had this changed in main but we decided it wasn't needed for Release branch - https://github.com/ni/measurement-plugin-labview/pull/656

Blahpapto1, 2025-08-19:
FYI: This is a known issue in VIPM. Just saw this in the repo - https://github.com/vipm-io/vipm-desktop-issues/issues/84

Tom-McQuillan, 2025-08-19:
Hello all,
Thank you to @Blahpapto1 for making me aware of this issue.

This is an unintended side effect of bug fix for another issue. I have captured this issue and have created unit tests to capture it.
The immediate fix for you is to disable the item in `Preview Features`. For the next patch release of VIPM, this will be disabled by default.

[image]

My apologies for the disruption to your workflow.

Please let me know if this resolves the immediate issue (unless you have already downgraded VIPM because of this).

Best regards,
Tom

Blahpapto1, 2025-08-20:
I couldn't find any options to update the v3.4.0.1 Example vip package in the VIPM Server. I've messaged Jim from JKI, waiting on his response.

closed 2025-08-22 by bharathirm11